dreemurrs-embedded/Jumpdrive #6: Good first impression README?

Danct12, 2020-03-17:
We probably should make one, so that they can get the first impression by reading the README.

Things we need:
- A banner: We can take the xcf from the splash screen, then add the word "Jumpdrive" next to it
- Better instructions on how to use it: So that they can find the utility easy to use.

Danct12, 2020-03-23:
Made a banner for Jumpdrive, feedback?

[image]

MartijnBraam, 2020-03-23:
white background with black text probably fits better in a readme

Danct12, 2020-04-24:
How about this then?

[image]

closed 2020-05-02 by Danct12