dreemurrs-embedded/Jumpdrive #7: panic: We're hanging here...

asriel-danctnix, 2020-03-17:
Meanwhile vast majority of users will not see this message *(assuming their eMMC and USB controller is working well)*, this might leave some user in confusion if their device is **defective**.

Especially the fact that some people will always read the last line, and only that, then complain in our DanctNIX/PinePhone chat room.

[image]

**Fun fact:**
FYI the "We're hanging here" message was from [Darwin/XNU kernel](https://github.com/apple/darwin-xnu) as one of our [original developer](https://github.com/Danct12) wanted that feel of a iOS device crashing, [the message looks like this on Apple systems](https://i.redd.it/klv3ow7nlczy.jpg)

Danct12, 2020-03-17:
Should've corrected that to "We are hanging here..."

https://github.com/apple/darwin-xnu/blob/a449c6a3b8014d9406c2ddbdc81795da24aa7443/bsd/kern/bsd_init.c#L966

MartijnBraam, 2020-03-17:
Replaced by a graphical error screen that displays the errormessage and then runs loop_forever so it can be debugged using the telnet shell

[image]

closed 2020-03-17 by MartijnBraam