dreemurrs-embedded/Jumpdrive #10: Connect to wifi + Expose NAND to network via SAMBA/NFS

droidmahn, 2020-03-21:
This is a feature request, and I imagine it'll be a bit of a pain and possibly out of scope, but I figured I'd try. 

I dont know about you guys but my pinephone's usb port is starting to get loose from all of the flashing. With things being what they are, it could be a while before we see replacement parts.

Being able to flash images over wifi would save our ports.
I propose a basic GUI to connect to wifi networks
and then exposing nand over NFS/Samba/SCP whatever works

MartijnBraam, 2020-03-22:
This seems way out of scope, this can't easily be done with the current built software (linux, uboot, busybox) those are easily compiled and don't have external dependencies except for a libc. That's why there's no SSH or samba/nfs server available. Also for wifi connection a whole GUI library is required.

Danct12, 2020-03-25:
We also don't want to bloat the image, as we want to keep it minimal as we could.

closed 2020-03-25 by MartijnBraam

droidmahn, 2020-03-26:
> We also don't want to bloat the image, as we want to keep it minimal as we could.

Just curious, but with micro sd cards smaller than 4gb being a rarity nowadays, why? Is the plan to eventually make this operate like a "recovery partition" on the emmc?

Is jumpdrive's scope more recovery based or just pure flashing?
If it's recovery, I'd personally prefer the max number of recovery tools possible to try and cover every scenario (emmc backups, shell, basic file manager (mc) etc) which would run counter to saving space.

swedneck, 2020-06-21:
heh, i literally can't find microSD cards smaller than 32GB in stores nearby, i personally don't feel that size is an issue..

MartijnBraam, 2020-06-22:
The big difference is that jumpdrive currently only has a kernel and busybox, this would require adding more components, which require a way more complex build system (since the kernel and busybox can be crosscompiled standalone)