dreemurrs-embedded/Jumpdrive #11: Boot selection when starting.

orencio, 2020-03-22:
Hello, is it possible that at boot time you can show options for booting from SD (with Jumpdrive) or eMMC (with the installes operating system)?

If it is, the sdcard can be always inserted in the pinephone, and when yoy want to flash anything you only must reboot.

Regards.

MartijnBraam, 2020-03-22:
There is a possibility to get megi's patched u-boot in here which supports reading the volume keys in the boot script. Then jumpdrive's boot.scr can check if the volume down key is held and launch the kernel/initramfs, if it's not held then load the boot.scr from emmc.

This will also probably break depending on what distro is currently flashed on eMMC since it might be using another uboot binary with ATF patches or might be using the extlinux.conf method for booting.

Goddard, 2020-03-27:
would be awesome if this could be provided as a package to distro developers when creating their pinephone images so they all can consistently integrate the functionality.

darkdragon-001, 2020-12-06:
One could just use megi's p-boot with only jumpdrive. This way one gets all the boot menu options and fast speed together with still quite a small size. This is what I am using myself.

grwlf, 2022-01-15:
Also having fun moving sdcard in and out repeatedly :-/ Will be glad to have a bootmenu or alike to alter the boot order