dreemurrs-embedded/Jumpdrive #25: Add an internal downloader/flasher

ScardracS, 2020-06-27:
Hi there.

I would suggest to add ability to connect to wifi in order to download and flash OS directly from jumpdrive.

In my idea we could make a list of "stock" images from every OS and add ability to download one of them and flash directly on eMMC, without passing from PC every time.

Please tell me if it possible,

Have a nice day,

Marco Scardovi (ScardracS)

ScardracS, 2020-06-27:
Obvs that's not an issue, just a suggestion

MartijnBraam, 2020-06-27:
That would involve UI for connecting to a network, a UI to select images, in a product that doesn't have any UI except splash screens. This is a bit out-of-scope.

ScardracS, 2020-06-27:
Understand. The idea was coming because often I haven't pc near, nothing more than that

MartijnBraam, 2020-06-27:
You should be able to flash the SD from the OS booted on eMMC and the eMMC from an os booted from SD. it might make sense to make an generic linux app for that that can be packaged in the available distros

ScardracS, 2020-06-29:
> You should be able to flash the SD from the OS booted on eMMC and the eMMC from an os booted from SD. it might make sense to make an generic linux app for that that can be packaged in the available distros

That could be an interesting feature

closed 2020-08-21 by ScardracS