dreemurrs-embedded/Jumpdrive #26: Unable to mount the exposed eMMC partition on Linux.

PoorPocketsMcNewHold, 2020-07-08:
```
               +                 User: pm
               #                 Hostname: VIX
              ###                Model: To Be Filled By O.E.M.
             #####               Distro: ArcoLinuxB-awesome [x86_64]
             ######              Kernel: 5.7.7-zen1-1-zen
            ; #####;             Uptime: 6 hours and 18 minutes
           +##.#####             WindowManager: awesome
          +##########            DesktopEnvironment: Not detected
         #############;          Shell: /bin/zsh
        ###############+         Terminal: Termite ## ## ## ## ## ## ##
       #######   #######         Packages: 1915
     .######;     ;###;`".       Temperature: 49.7 C (Max. 66.2 C)
    .#######;     ;#####.        CPU: AMD Ryzen 5 3600X 6-Core Processor
    #########.   .########`      GPU: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1)
   ######'           '######     RAM: 6448 MiB / 32042 MiB
  ;####                 ####;    Disk: 4655.7 GiB / 6101.4 GiB
  ##'                     '##    LAN_IP: 192.168.1.38
 #'                         `#   WAN_IP: ##.###.##.###
```

After downloading the latest 0.5 release available from GitHub and flashing the image onto my ext4 partitioned 64 SD Card. (Big, but what I’m currently working with) via dd.

\`sudo dd if=pine64-pinephone.img of=/dev/sdh bs=1M status=progress conv=fsync\`

I put it into my Pinephone CE UBPorts, and it correctly boots into the Jumpdrive screen as expected.

However, It doesn’t show up at all on my computer, with the default Pinephone red cable, having tried using classic USB 2.0 and 3.0 available on the front panel of my desktop.

Nothing interesting displaying there. 

```
sudo cat /sys/kernel/debug/usb/devices | grep -E "^([TSPD]:.*|)$"                                                                                                                                       14:46:21 

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh=14
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.07
S:  Manufacturer=Linux 5.7.7-zen1-1-zen xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:02:00.0

T:  Bus=01 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  2 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=03eb ProdID=ff01 Rev= 0.50
S:  Manufacturer=Wooting
S:  Product=WootingOne
S:  SerialNumber=WOOT_001_A255B1801W010H00516

T:  Bus=01 Lev=01 Prnt=01 Port=04 Cnt=02 Dev#=  3 Spd=480  MxCh= 3
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0409 ProdID=005a Rev= 1.00

T:  Bus=01 Lev=02 Prnt=03 Port=00 Cnt=01 Dev#=  6 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=ff(vend.) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=045e ProdID=02ad Rev= 1.00
S:  Manufacturer=Microsoft
S:  Product=Xbox Kinect Audio, © 2011 Microsoft Corporation. All rights reserved.
S:  SerialNumber=A44866B02710038A

T:  Bus=01 Lev=02 Prnt=03 Port=01 Cnt=02 Dev#=  5 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=045e ProdID=02b0 Rev= 1.07
S:  Manufacturer=Microsoft
S:  Product=Xbox NUI Motor

T:  Bus=01 Lev=02 Prnt=03 Port=02 Cnt=03 Dev#=  7 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=045e ProdID=02ae Rev= 1.0b
S:  Manufacturer=Microsoft
S:  Product=Xbox NUI Camera
S:  SerialNumber=A00361901557039A

T:  Bus=01 Lev=01 Prnt=01 Port=05 Cnt=03 Dev#=  4 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=03f0 ProdID=e207 Rev=10.22
S:  Manufacturer=Hewlett Packard
S:  Product=HP Webcam HD 2300

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 8
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.07
S:  Manufacturer=Linux 5.7.7-zen1-1-zen xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:02:00.0

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.07
S:  Manufacturer=Linux 5.7.7-zen1-1-zen xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:0d:00.3

T:  Bus=03 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1b1c ProdID=1b3c Rev= 3.08
S:  Manufacturer=Corsair
S:  Product=Corsair Gaming HARPOON RGB Mouse
S:  SerialNumber=0E04101CAF2D80C55AE2608CF5001BC1

T:  Bus=03 Lev=01 Prnt=01 Port=01 Cnt=02 Dev#=  3 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=28de ProdID=1142 Rev= 0.01
S:  Manufacturer=Valve Software
S:  Product=Steam Controller

T:  Bus=03 Lev=01 Prnt=01 Port=03 Cnt=03 Dev#=  4 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0a12 ProdID=0001 Rev=88.91
S:  Product=CSR8510 A10

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 4
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.07
S:  Manufacturer=Linux 5.7.7-zen1-1-zen xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:0d:00.3
```

PoorPocketsMcNewHold, 2020-07-08:
@Danct12 pointed me out (https://canary.discordapp.com/channels/463237927984693259/543382344782970881/730407743252463688) that UBPorts already had Jumpdrive built-in. So, I’ve removed the SD card with tested with that one instead, but it didn’t changed anything.

PoorPocketsMcNewHold, 2020-07-08:
Manage to make it show on a Zorin OS laptop. I’ll take that this was an issue on my end. I’ll investigate this myself then.

KylePiira, 2020-07-10:
I was also having this issue on Linux Mint. I tried v0.5 but it never mounted the eMMC. After reverting backt o v0.4 it worked.

droidmahn, 2020-07-13:
Same issue on Arch Linux. Tried on 2 different machines so far ill try reverting to 0.4

PoorPocketsMcNewHold, 2020-07-14:
Re-opened it, since at the end. We don’t really know if we suffer from the same issue, what dependency is missing for making us detect the exposed bootloader.

muellermartin, 2020-07-23:
I have the same issue on macOS (the eMMC partition of a PinePhone Braveheart edition is not exposed via USB). I was able to work around this issue by using JumpDrive 0.4 instead of JumpDrive 0.4.

Danct12, 2020-07-29:
Not really too sure why is that happening here, I've tested 0.5 with my PinePhone 1.2a and 1.0, everything seems to work fine there.

Gopherslol, 2020-08-06:
I am getting the same issue on my ThinkPad W700 running Manjaro. I tried both 0.5 and 0.4. It worked on my T530 running Solus. Oddly enough, it does not appear to be OS dependent because I can replicate the issue on my desktop running Solus

Danct12, 2020-08-09:
Does anyone here have UART cable? If so, please get dmesg logs when the phone is plugged in, and unplugged.

gudenau, 2020-08-11:
I don't have a debug cable but I am having the same issue on two computers across three USB cables and 5 USB ports. The USB ports being Type-C, USB A 2 and USB A 3.

reivanen, 2020-08-19:
I seem to have this issue, both my desktop and laptop (manjaro and ubuntu) do not detect any drives. 'dmesg is spammed with errors like this:

[ 4328.264446] usb 1-6: new full-speed USB device number 11 using xhci_hcd
[ 4328.264517] usb 1-6: Device not responding to setup address.
[ 4328.471198] usb 1-6: Device not responding to setup address.
[ 4328.677758] usb 1-6: device not accepting address 11, error -71
[ 4328.677827] usb usb1-port6: unable to enumerate USB device

Danct12, 2020-08-20:
@reivanen Can you tell me what Jumpdrive version? And have you tried another USB cable?

reivanen, 2020-08-20:
i believe it was 0.5.0 the download naming scheme does not tell. Here is the URL:

[link](https://github-production-release-asset-2e65be.s3.amazonaws.com/247024339/7f1c1980-bb05-11ea-968e-5e9c5887870d?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200819%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200819T130050Z&X-Amz-Expires=300&X-Amz-Signature=04c5f53d21334a7c7b58a2d0cdb90f278ac7155439df7da9b779b2e0a81362a9&X-Amz-SignedHeaders=host&actor_id=0&repo_id=247024339&response-content-disposition=attachment%3B%20filename%3Dpine64-pinephone.img.xz&response-content-type=application%2Foctet-stream)

reivanen, 2020-09-02:
I tried again, this time using a 64gb memory card and burning the image using ubuntu's integrated image burner. (vs. 512gb card and etcher) and it seems to be working correctly.

magedragon25, 2020-11-09:
I tried with all versions post .4, different computers and different linux distros with no success to mount emmc. Only when I used ver .4 on my arch system was I able to mount the exposed emmc. Currently flashing SailFish OS to the pinephone emmc as I type this.

Ba1238, 2020-11-17:
Just make sure your battery is charged. If it is empty emmc is not recognized.

PoorPocketsMcNewHold, 2020-11-25:
So, update on that. I’ve managed to fix it!

The thing is, It seems that the stock pinephone red cable I’ve been using, seems to only be good for energy transfer, and not being able to do data transfer.

I’ve switched to another cable which have been using with other androids phone for modding as well, and low and behold, I’ve got access to the eMMC!

Despite knowingly that other people seem to have the same issue, I’ll close that issue again for now, waiting for confirmation, or not from them.

closed 2020-11-25 by PoorPocketsMcNewHold

chmol, 2020-11-29:
I had this weird behaviour:

couldn't see the storage or telnet until I've removed the battery.  Using the pine64 supplied cables (only have that).