dreemurrs-embedded/Jumpdrive #27: Support the 3GB pinephone

MartijnBraam, 2020-08-15:
U-boot needs a newer release to support the 3GB memory modules in the convergence edition pinephone

MartijnBraam, 2020-08-15:
Having some issues with applying the 3GB patch for u-boot, right now u-boot crashes with 2020.07 + the existing patch + the patch from icenowy for 3GB

```
DRAM: 4096 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.2(release):v2.2-904-gf9ea3a629
NOTICE:  BL31: Built : 15:32:12, Apr  9 2020
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x4063418, model: PinePhone
NOTICE:  PSCI: System suspend is unavailable

U-Boot 2020.04 (Aug 15 2020 - 18:12:56 +0200)

DRAM:  3 GiB
"Synchronous Abort" handler, esr 0x86000000
elr: 1100000004000000 lr : 1100000004000000
x0 : 0000000030c50830 x1 : 0000000000000008
x2 : 000000004a063418 x3 : 000000004a063418
x4 : 00000000fffb9bf0 x5 : 0000000000017404
x6 : 0000000000000000 x7 : 0000000000000000
x8 : 0000000000017888 x9 : 00000000b5f75000
x10: 000000004a044bf0 x11: 000000004a001850
x12: 0000000000000006 x13: 000000000001869f
x14: 000000004a063418 x15: 0000000000000000
x16: 000000004a02af10 x17: 0000000000000000
x18: 00000000fbf54df0 x19: 0300000001000000
x20: 0000000000000000 x21: 0000000000000000
x22: 0000000000000000 x23: 0000000000000000
x24: 0000000000000000 x25: 0000000000000000
x26: 0000000000000000 x27: 0000000000000000
x28: 0000000000000000 x29: 0300000000000073

Code: "Synchronous Abort" handler, esr 0x96000000
elr: 000000004a001f1c lr : 000000004a001efc
x0 : 000000004a04a13e x1 : 0000000000000000
x2 : 0000000000000020 x3 : 0000000001c28000
x4 : 0000000001c28000 x5 : 00000000fbf4e746
x6 : 0000000000000033 x7 : 000000000000000f
x8 : 00000000fbf4ebc8 x9 : 00000000b5f75000
x10: 00000000ffffffe8 x11: 0000000000000010
x12: 0000000000000006 x13: 000000000001869f
x14: 000000004a063418 x15: 0000000000000021
x16: 000000004a02af10 x17: 0000000000000000
x18: 00000000fbf54df0 x19: 1100000003fffff0
x20: 000000004a04e441 x21: 00000000fffffffc
x22: 000000004a04ee6b x23: 000000004a04a13e
x24: 0000000000000000 x25: 0000000000000000
x26: 0000000000000000 x27: 0000000000000000
x28: 0000000000000000 x29: 00000000fbf4ec00

Code: d1004273 9139aed6 aa0003f7 12800075 (b9400261) 
Resetting CPU ...

resetting ...
```

josteink, 2020-09-01:
I'm getting the PinePhone 3GB convergence edition any time now. Any way I can assist testing this issue?

1001111github, 2020-09-03:
Just tested 0.6 alpha on the 3GB CE edition. It has worked so far without any issues.

MartijnBraam, 2020-09-04:
Seems like the 0.6 build breaks on braveheart though

josteink, 2020-09-04:
The change involved seem to introduce a new dtb. Could that be it?

https://github.com/dreemurrs-embedded/Jumpdrive/commit/829a574635f85fcbe2b8a9486e4156298b2c9e18

yoursunny, 2020-09-14:
I tried `0.6-alpha` release on PinePhone postmarkOS 3GB edition.
The phone turns on and displays the logo and telnet instruction.
However, when connected to a computer (I tried a BeagleBone Black and a Raspberry Pi 4 and a Windows 10 laptop), **no USB device shows up** (`lsusb` has no additional device).
I'm using the cable that comes with the package. I tested this cable with another USB-C phone and `lsusb` sees a device, which means the cable has data pins.

dcolt, 2020-09-16:
> I tried `0.6-alpha` release on PinePhone postmarkOS 3GB edition.
> The phone turns on and displays the logo and telnet instruction.
> However, when connected to a computer (I tried a BeagleBone Black [...]
I'm having the same issue. Also an postmarketOS 3GB edition

josteink, 2020-09-16:
For me, it sometimes seems to matter which USB-port I insert it into on the computer. 

I'm having the same issue with some cheap Chinese USB serial adapters. For those, using plain old USB2.0 port on some hub worked best.  

That said, I haven't gathered any data on which ports works best for Jumpdrive yet.

yoursunny, 2020-09-16:
I don’t own a HUB.
The single USB port on BeagleBone Black is a USB 2 port. I have used this port to flash many other devices.
The USB ports I used on Raspberry Pi 4 or Windows laptop are both USB 3.

mckenzielong, 2020-09-20:
Also used 0.6 alpha, on a PostMarket OS community edition 3gb model. Couple bumps, but eventually worked on both usb 2 and 3 ports.

The PinePhone usb-c cable refused to work for me - both USB 2 and 3 ports. I ended up having to use a different usb cable entirely. Included dock also didn't want to place nice. Just a regular high quality usb c cable did the trick.

tomatic, 2020-09-22:
Tried it on two separate machines, different USB ports and cables, with and without the hub, no dice. The phone just does not show up, lsusb doesn't list it.

a-v-e-s, 2020-09-28:
I just wanted to comment that 0.6 Alpha worked for my 3 GB Convergence Edition Pinephone as well, while other versions failed. Thanks to everyone who contributed!

m0jek, 2020-10-07:
Thank you for 0.6 Alpha, works for me too on my 3 GB CE Pinephone, Phone connected to macOS just fine too.

thekix, 2020-10-09:
Hi,

I had the same problem, when I connect the device nothing happens. My procedure:

1. flash the sd card and install it in the phone
2. put the battery
3. boot the phone (do not connect it), wait for the jumpdrive image on the screen
4. connect the cable to the phone and connect to the computer. I used a 2.0 USB port. Nothing happens.
5. disconnect the phone and wait 2 seconds
6. connect the cable again to the computer

```
[92330.101902] usb 1-11: new high-speed USB device number 6 using xhci_hcd
[92330.339558] usb 1-11: New USB device found, idVendor=1209, idProduct=4201, bcdDevice= 5.07
[92330.339561] usb 1-11: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[92330.339562] usb 1-11: Product: PinePhone
[92330.339563] usb 1-11: Manufacturer: Pine64
[92330.339564] usb 1-11: SerialNumber: Jumpdrive
...
[92331.411126] sd 10:0:0:0: [sdf] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[92331.411250] sd 10:0:0:1: [sdg] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[92331.437619]  sdf: sdf1 sdf2
[92331.442389]  sdg: sdg1
[92331.442801] sd 10:0:0:0: [sdf] Attached SCSI removable disk
[92331.445112] sd 10:0:0:1: [sdg] Attached SCSI removable disk
```

Cheers,
kix

sokolgeo, 2020-11-02:
Unable to connect 3 GB Convergence Edition Pinephone using 0.6 Alpha.
Tried on two linux boxes, different USB ports, two different USB type C cables.
Either nothing in dmesg or unable to read USB device descriptor error.
Overall - does not work.

MartijnBraam, 2020-11-08:
I pushed 0.6 Alpha 2 now, maybe that fixes it?

josteink, 2020-11-08:
> I pushed 0.6 Alpha 2 now, maybe that fixes it?

Tested on my pmOS convergence-edition PinePhone and it works fine :+1:

Wivik, 2020-11-12:
Hello,
I've just flashed my Pinephone (acquired from the last dispatch wave) from the original Manjaro to PostmarketOS with the 0.6 release. Seems fine.

A first, the eMMC didn't mount but it seems to be related to the bundled cable. I've tried with another USB-C cable and it was OK.

Here is how I proceed : 
- Flashed jumpdrive pinphone to sdcard
- Inserted sdcard on pinephone
- Plugged in the USB-C cable, it switched on the phone, Jumpdrive popped on screen
- Filesystems showed up  with new cable, nothing with original one (`lsusb` and `lsblk` were silent)

yoursunny, 2020-11-14:
v0.6 is working for me.

* 3GB postmarketOS edition
* original cable
* battery must be inserted to the phone; without a battery, the copying stalls after 5 minutes
* `bs=1M conv=fsync` must be specified; without these flags, the copying fails with I/O error after 20 seconds

```shell
$ xzcat Manjaro-ARM-phosh-pinephone-beta2.img.xz | sudo dd of=/dev/sdb bs=1M conv=fsync status=progress
0+71288 records in
0+71288 records out
4671406080 bytes (4.7 GB, 4.4 GiB) copied, 9294.65 s, 503 kB/s
```

Bitbooger, 2020-11-16:
For everyone who is having issues, I made a video on flashing Mobian (no difference to other OSs of course) to the eMMC with Jumpdrive with the new 3GB Manjaro CE model. Maybe you made a tiny mistake (forgetting to "sync" after dd [like I did on my first attempt, screen stayed black] or something like that) that you can eliminate watching the video. I don't want to make a commercial here, I don't earn any money at this point in time with my YouTube channel, I just want to help people who are struggeling or who need some guidance. Also, please read the description of the video prior to watching it, I wrote an important note about unmounting the device before running dd command.

Hope it helps: https://www.youtube.com/watch?v=et9IQt6uxkg

closed 2020-12-23 by Danct12

steils, 2021-06-16:
I don't know if it's my fault, but just for the record. I have a 3GB PinePhone, using jumpdrive-0.8. It starts, shows the splash screen, but I don't see any devices on `lsusb` or `lsblk`, and there is silence in dmesg.
I'll try to debug it later.