dreemurrs-embedded/Jumpdrive #31: Support for PineTab Early-Adopter Edition

jkoan, 2020-09-17:
Hi folks,

currently Jumpdrive on the PineTab Early-Adopter Edition has broken Display support.
It only shows up as black. This could probably be due to the [LCD panel driver swap from pine64 ](https://www.pine64.org/2020/07/24/all-about-the-pinetab-update/)
USB-gadget and Telnet are working without any problem.

Tested with Jumpdrive 0.5

myxor, 2020-09-22:
I can confirm this on my PineTab Early Adopters Edition as well.

MartijnBraam, 2020-11-08:
Pushed a new alpha build that might solve this (I don't have that model pinetab to verify it)

jkoan, 2020-11-08:
Its Working, thx

Danct12, 2020-11-09:
Now a part of [Jumpdrive 0.6](https://github.com/dreemurrs-embedded/Jumpdrive/releases/tag/0.6)

closed 2020-11-09 by Danct12