dreemurrs-embedded/Jumpdrive #32: Support preloading JumpDrive with an image

faissaloo, 2020-10-21:
Given some of the early PinePhones can't communicate over USB without soldering things and to make it easier for distros to distribute installable images it'd be cool to be able to preload Jumpdrive with the image you want to write to eMMC before writing it to an SD card rather than having to send it over USB.

MartijnBraam, 2020-10-21:
The early pinephone batches without usb fix still work fine with JumpDrive, it's using usb gadget mode. The thing that's broken is host mode switching without the fixes that require soldering.

faissaloo, 2020-10-22:
Oh awesome thanks!

closed 2020-10-22 by faissaloo

NoahAndrews, 2020-12-17:
In my opinion this is still a useful feature, because it would allow switching OSes without having to use an external computer.