dreemurrs-embedded/Jumpdrive #38: Cannot enable. Maybe the USB cable is bad?

Aurze, 2021-01-10:
I've tried with multiple cable to plug my pinephone (3G) but I don't see my device and I always get the following logs in dmesg when I plug it in.
```
[120908.316913] usb usb1-port4: Cannot enable. Maybe the USB cable is bad?
[120908.316949] usb usb1-port4: unable to enumerate USB device
```

Device was originally a Ubports edition but the main board was upgraded to the 3G edition.

the-maldridge, 2021-02-28:
I'm getting the same error, were you able to resolve this @Aurze ?

Grzesio-ops, 2021-03-17:
@Aurze @the-maldridge I got the same problem, fixed it by connecting phone to usb ports on back of my pc instead of front of it.

Aurze, 2021-03-25:
> I got the same problem, fixed it by connecting phone to usb ports on back of my pc instead of front of it.
That did fixed my issue.
Also sometime restarting the computer with my phone plugged in help.

Aurze, 2021-05-06:
Jump-drive doesn't seems to charge the device. Are you sure your batterie is over 50%?