dreemurrs-embedded/Jumpdrive #41: Installed Jumpdrive 0.6 on SD Card, but Screen stays black and LED is blinking Green every second

GregoriusT, 2021-02-02:
I tried charging the Battery, but it doesn't seem to create a different reaction.

The Screen stays black (not even backlit) in all cases, while the indicator LED is flashing green at one Hertz.

I watched a random Tutorial Video on this, and noticed that there is definitely supposed to be Screen activity going on. So something is wrong enough for me to report this.

When I plug it into my Computer, not even dmesg is reacting to it, as if it doesn't exist. I tried all my other USB-C Cables (including a super short one) and nothing is being recognized. Using other USB Ports also does not help, not even plugging it into my Raspberry Pi 4B helped (neither the USB 3 Ports nor the USB 2 Ports).

craftyguy, 2021-02-10:
I see this sometimes, usually re-flashing it on the SD card (make sure to sync before removing it...), or trying a different SD card 'resolves' the issue for me.

ntfc, 2021-02-10:
Had the same thing as OP, and in my case I was wrongfully flashing the `.img.gz` instead of the `.img` file to the SD card 🤦

GregoriusT, 2021-02-10:
I did flash an .img File after using unxz to extract it, so I guess I'm gonna try re-flashing. There should be something about this Issue in the ReadMe, that tells about Blackscreen with Green Indicator LED. ;)

Also is it intended that image.gz and initramfs.gz are in the SD card?

GregoriusT, 2021-02-10:
Great, now the Phone behaves as if the Battery doesn't exist...