dreemurrs-embedded/Jumpdrive #48: [Suggestion] Add PinePhone modem firmware flash tools

PReDiToR, 2021-04-16:
To give Biktorg and other community members a standard environment to do firmware flashing with a known OS and toolset, having a button on JumpDrive that drops to a shell to run the install or recovery scripts may be helpful. Even an ncurses or bash menu to automate it.

Having a full stack for networking is beyond the scope of my suggestion, but having a FAT32 partition on the end of the image to copy latest files to may be very little extra work.

Danct12, 2021-04-17:
There's no way this tool can be distributed **legally**, as the qfirehose utility provided by Quectel is proprietary.

Danct12, 2021-04-17:
CC: @MartijnBraam, what do you think about this?

MartijnBraam, 2021-04-17:
Out of scope

aroggeband, 2022-07-22:
No need to ship firmware or propietary tools. Too many problems, too much maintenance. Just a stable install of android-tools is a good starter kit.

Danct12, 2022-07-22:
TBH, I don't see a reason why we should add this. People can just boot to a live system (or a Arch Linux Barebone install), download everything needed and install/patch the modem.

Can we please keep this as a simple recovery tool for devices and not a recovery tool for everything connected?