dreemurrs-embedded/Jumpdrive #49: Add CONFIG_MMC_BROKEN_CD to U-Boot config

8bitgc, 2021-05-09:
This may allow PinePhones with damaged SD sockets to boot off SD.

Damaged card detect switch in microSD socket causes U-Boot to halt with:
```
Trying to boot from MMC1
MMC: no card present
spl: mmc init failed with error: -123
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
```
It fails even though U-Boot is running from the SD card itself. mmc_start_init tries to initialize the card and fails on the card detection.

https://gitlab.com/pine64-org/u-boot/-/blob/master/drivers/mmc/mmc.c#L2867-2879

CONFIG_MMC_BROKEN_CD should allow U-Boot to continue ignoring the state of the card detection pin.

Some background information in this post https://forum.pine64.org/showthread.php?tid=13775

8bitgc, 2021-05-17:
This seems to work. Compiled with CONFIG_MMC_BROKEN_CD and tested on a PinePhone with a broken card detection switch. https://github.com/8bitgc/Jumpdrive/tree/master/test

Danct12, 2021-05-17:
Reopening the issue as this needs to be implemented to Jumpdrive.