dreemurrs-embedded/Jumpdrive #55: Undocuemented pine64-pinephone-charging.img.xz

kkofler, 2021-07-23:
The 0.8 release contains a new file `pine64-pinephone-charging.img.xz`. Unfortunately, I could not find any documentation for it, and even browsing through the source code, I do not see it mentioned anywhere. Hence my questions: What is that file? How is it generated? And perhaps most importantly, how does it differ from the regular `pine64-pinephone.img.xz` and which one do I want?

MartijnBraam, 2021-07-24:
It's an image for testing, I didn't want to push a new tag because then there'd be complaints that the new release is missing most images.

It's a build with a different kernel config and initramfs that disables the display of the PinePhone and uses the RGB led to show the charging state by blinking different colors. It blinks red if the battery is <5%, it blinks yellow when it's between 5 and 50% and it blinks green when it's above 50%

This is specifically to deal with pinephone batteries that are so low that a normal OS won't boot to charge it. I'll push the changes to a different branch since it's a mess.

MartijnBraam, 2021-07-24:
Pushed to https://github.com/dreemurrs-embedded/Jumpdrive/tree/feature/charging

kkofler, 2021-08-20:
Even if they changed that now, that would not magically fix all the hardware out there, would it?

kkofler, 2021-08-20:
(And as far as I know, they do not really plan new hardware revisions to begin with. I think they want to stick to 1.2 now, even if and when they remove the Beta label. Though I may be wrong, since I do not have any contacts inside Pine64.)