dreemurrs-embedded/Jumpdrive #56: Can't boot into Jumpdrive on PinePhone

dave-kennedy, 2021-08-13:
I'm not sure what I'm doing wrong here:

```
$ lsblk
sda           8:0    0 223.6G  0 disk 
├─sda1        8:1    0    16M  0 part 
├─sda2        8:2    0   223G  0 part 
└─sda3        8:3    0   517M  0 part 
sdd           8:48   1   1.9G  0 disk 
nvme0n1     259:0    0 238.5G  0 disk 
├─nvme0n1p1 259:1    0   487M  0 part /boot/efi
└─nvme0n1p2 259:2    0   238G  0 part /

$ ls -l pine64-pinephone.img.xz 
-rw-rw-r-- 1 dave dave 5541868 Aug 12 18:10 pine64-pinephone.img.xz

$ sudo dd if=pine64-pinephone.img.xz of=/dev/sdd bs=1M status=progress conv=fsync
5+1 records in
5+1 records out
5541868 bytes (5.5 MB, 5.3 MiB) copied, 3.39768 s, 1.6 MB/s
```

When I put the card in the phone and turn it on, it boots to Plasma Mobile desktop. I've tried two different SD cards. Am I missing something?

dave-kennedy, 2021-08-13:
I got it to boot, finally. This time I made sure to `eject /dev/sdd` before removing the drive. It might help others if the readme mentioned any of this.

warrenc5, 2021-09-05:
`xzcat pine64-pinephone-charging.img.xz | dd of=/dev/sdf bs=1M status=progress conv=fsync`

jmevel, 2024-10-30:
@warrenc5 

Hi, I screwed up my pinephone (the old non-pro one) eMMC today (flashed it with `dd` while I was actually booted on it... yeah stupid mistake)

So I'm trying to flash it again with Jumpdrive. 
I first tried to flash `pine64-pinephone.img` the LED was turning on but nothing else. Then I tried `pine64-pinephone-charging.img`, now the LED is simply blinking green but nothing else as well. 

Any idea what's going on? What's the difference between `pine64-pinephone.img` and `pine64-pinephone-charging.img`?

I'm a bit desperate now, I'm afraid I won't be able to fix my phone if I can't even make Jumpdrive to start...

Thanks for your answer