dreemurrs-embedded/Jumpdrive #60: Jumpdrive flashed to microSD and showing prompt on pinephone, but not allowing connection to eMMC or microSD card once connected

User10987654321, 2021-09-15:
I successfully dd'd jumpdrive to a microSD, which I then inserted into a pinephone that I plugged into a usb port on the computer. The jumpdrive screen-prompt appears on the phone only once then phone is connected to the computer, indicating the problem is not with the pinephone charging port, the pc usb port, or the usb cord that connects the two.

Before this problem, I had jumpdrive working once (meaning the drive appeared in the file manager, but I pressed eject, a function that 'erases' the drive from the current usb connection, and always allows for usb flash drives to reconnect successfully, being how I've reinstalled jumpdrive so many times as well as dd'd it with /dev/zero numerous times.). The problem now is that after the ejection, the pinephone hasn't been recognized once with jumpdrive.

Despite this, no drives appear in /dev, xfce thunar, or balena etcher that could possibly be the pinpehone when I connect it with the jumpdrive-microSD card in the pinephone with the jumpdrive prompt on the pinephone the entire time.

How can I connect the pinephone with jumpdrive to get it recognized by the pc, which it is currently unable to accomplish for the above reasons.

slackerbob, 2021-11-03:
I have no idea if this is a real fix or just coincidence, but it seems like the jump drive only works for me when the battery is removed (and left out) before booting.

bam80, 2021-11-17:
@slackerbob what is the hw revision of your pinephone?
I wonder if it makes sense to try on Braveheart.

slackerbob, 2021-12-02:
@bam80 I haven't figured out how to find that info without taking out the mainboard, but I only purchased it in the last few months, so it should be the newest revision.