dreemurrs-embedded/Jumpdrive #61: Please add cryptsetup

Silent-Hunter, 2021-09-17:
This would be very usable for repairing an installation with FDE enabled if it had cryptsetup. My Pinephone froze during a PostmarketOS update, and wouldn't boot afterward. I was able to fix it using a Raspberry Pi 4 with the experimental 64-bit Raspberry Pi OS, but I would have been able to do it with Jumpdrive and my regular PC if Jumpdrive had cryptsetup and chroot capabilities.

MartijnBraam, 2021-09-19:
it's not for jumpdrive to implement, if you have cryptsetup on your host PC it Just Works(tm)

Silent-Hunter, 2021-09-20:
But then I can't chroot because the architecture of my PC is not the same as the architecture of the Pinephone.
‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

Danct12, 2021-09-20:
FYI, binfmt-qemu-static exists as a way to run binaries that are different from your architecture.

Silent-Hunter, 2021-09-20:
Oh, I know. I thought it would be easier if it was part of Jumpdrive, but I understand if it's too big or bloated or something.
‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2021-09-26 by Danct12