dreemurrs-embedded/Jumpdrive #63: PinePhone is not detected as USB device by the PC

bam80, 2021-11-17:
While "Jumpdrive is running" prompt on the screen, PC still doesn't see the device and it's not listed in `lsusb`.
I suspect that's maybe because of buggy hw revision of my phone, how can I check for sure which one I have?

wzuo, 2021-11-21:
I have the same problem. Running Kubuntu 20.10

[1416255.329694] usb 1-4: new full-speed USB device number 24 using xhci_hcd
[1416255.329884] usb 1-4: Device not responding to setup address.
[1416255.537641] usb 1-4: Device not responding to setup address.
[1416255.745784] usb 1-4: device not accepting address 24, error -71

nmauersberg, 2022-02-28:
Same here, been stuck on this for a while now..

bam80, 2022-03-05:
Whoever experiencing this problem please try to detect your hw revision.

gabrieldevillers, 2022-08-22:
Apologies for being trivial, but this may "simply" be hardware problem: I have a PinePhone with a damaged USB-C connector that behaves exactly like that (same messages), unless a small pressure is applied on the cable (such as having the phone lay on the cable end). I would suggest trying to change the [USB-C sideboard](https://pine64.com/product/pinephone-usb-c-side-board/) (currently out of stock :disappointed: ) and maybe also the [FPC cable](https://pine64.com/product/pinephone-fpc-cable/).

sven-frotscher, 2023-10-06:
> Whoever experiencing this problem please try to detect your hw revision.

Can you do that in the Jumpdrive shell? If so, how?