dreemurrs-embedded/Jumpdrive #66: README does not explain what `-charging` image name means

ell1e, 2022-01-14:
The README does not explain what `-charging` image name means. I think it should. Does that image have an entirely different purpose? Is it the same, but it also supports charging? If the latter, I think you should possibly add a misc folder, then put in the NON-charging image there, and just have the full charging image in the default list. It's tiny anyway, and I don't see why people would not want that functionality, so splitting it like that is a bit confusing.

ell1e, 2022-01-14:
Ok, chat told me that the normal image already does charging. I think the least to be done would be to rename the `-charging` image to `-chargingonly` to make that more obvious. Also, maybe that image should display a short notice on the screen when it boots what it's actually doing - I was told the screen is off to speed up charging, but surely displaying a quick "Charging now, screen will turn off to increase speed" notice wouldn't make such a huge energy difference while it would make it endlessly less confusing what the image even does when booted.

tpiekarski, 2022-02-26:
That's hilarious, I thought this -charging image would be to use Jumpdrive while a PinePhone is charging, because the README mention issues while PinePhone is charging.

> WARNING: In case eMMC isn't showing up try to charge your battery and try again. There are reports of people not being able to flash to eMMC with discharged battery.

This -charging image is only meant for charging and not booting Jumpdrive? 
If so than that figures why no boot screen - no nothing happens... only charging and I was wondering why that PinePhone won't start.

I definitely back this Feature Request and would like to provide a PR :-)

macias, 2022-02-27:
Related: https://github.com/dreemurrs-embedded/Jumpdrive/issues/55 Undocumented pine64-pinephone-charging.img.xz