dreemurrs-embedded/Jumpdrive #67: Self-built image doesn't work (pinephone)

Zapeth, 2022-01-23:
I tried to manually build an image for pinephone on void linux (glibc) with the distribution-provided cross-compiler (cross-aarch64-linux-gnu).

Compilation was successful (after resolving some build errors due to missing packages), however when trying to boot the image on pinephone, nothing happens (no led, no screen).

With the pre-built image the phone successfully boots, what could be the issue?

arbruijn, 2022-01-30:
The image works for me after building on void linux (packages: git base-devel wget cross-aarch64-linux-gnu cpio uboot-tools u-boot-tools mtools python3 swig python3-devel parted).
The build system doesn't handle restarts after failure so well though, did you try deleting the directory and cloning again after installing all required packages?

You should be able to mount the image with something like: mount -o loop,offset=1048576 pine64-pinephone.img /mnt/tmp
Does that work?

Zapeth, 2022-01-31:
> The image works for me after building on void linux (packages: git base-devel wget cross-aarch64-linux-gnu cpio uboot-tools u-boot-tools mtools python3 swig python3-devel parted).
> The build system [...]

Yes, I recloned the repo after figuring out which packages were missing (I directly installed the required packages instead of `base-devel`, but I don't think that should matter).

> You should be able to mount the image with something like: mount -o loop,offset=1048576 pine64-pinephone.img /mnt/tmp
> Does that work?

Structure-wise the created image was identical to the one from the release so yes, I was able to mount it directly and the files looked fine as well.

After some more troubleshooting I was actually able to boot my image by replacing the u-boot parts of the image file with the ones from the release, so I guess the issue is there somewhere.

Out of curiosity I tried building with a newer u-boot version (removing [u-boot-pinephone.patch](https://github.com/dreemurrs-embedded/Jumpdrive/blob/master/src/u-boot-pinephone.patch) in the process) and that worked finally, so I guess this issue can be closed again.

closed 2022-01-31 by Zapeth