dreemurrs-embedded/Jumpdrive #68: Pinephone Pro support

BurnyBoi, 2022-01-27:
The latest JumpDrive release for Pinephone doesn't boot on Pinephone Pro, correct? If so, consider this a request please!

Joshndroid, 2022-01-31:
I would also +1 for this please. Stock build for me is buggy and i need to reinstall it.

AsHperson, 2022-02-01:
Looking forward to this as I'd like to flash Manjaro with Phosh!

mossymossmoss, 2022-02-01:
I'm planning to use the PPP as my daily driver, having my OS flashed to EMMC would definitely improve the experience

TheKrafter, 2022-02-02:
Yes this would be great. The Mobian Project said it only took about an hour to port, and thats with a whole OS. https://blog.mobian-project.org/posts/2021/12/28/pinephone-pro/

arbruijn, 2022-02-03:
As an experiment I did some initial work on the Makefile. It builds an image, but I can't test it without a device. Probably it needs changes in the kernel/u-boot config/version.
https://github.com/arbruijn/Jumpdrive/tree/pinephonepro

lllsondowlll, 2022-02-08:
> As an experiment I did some initial work on the Makefile. It builds an image, but I can't test it without a device. Probably it needs changes in the kernel/u-boot config/version. https://github.com/arbruijn/Jumpdrive/tree/pinephonepro

If you compile a binary image and publish it to your releases I would be more than happy to test it.

mossymossmoss, 2022-02-13:
> While it's not ideal, you can just boot any OS from SD card, and use this OS to flash any image you want to eMMC, without the need for Jumpdrive

I've considered doing that, but some people have advised against it, saying that it's unsafe and that I should just wait for JumpDrive to be ported

rayrrr, 2022-03-04:
@arbruijn I have the device, but I am new to compiling OS binaries, especially for embedded. I followed the build instructions in the README on your fork's branch for this feature (same instructions as here) and it resulted only in a `src/linux-ppp` directory containing about 1.1GB of files. Where would I go from there to produce a binary image, or am I on the wrong track? Thanks.

PureTryOut, 2022-03-07:
Note that almost definitely JumpDrive is not going to be ported to the PinePhone Pro due to [Tow-Boot](https://tow-boot.org) existing with it's own JumpDrive-like mode by holding the volume up when booting till the led lights up blue and thus there being no need for JumpDrive. Please install Tow-Boot instead, it's a better booting experience anyway! @Danct12 please confirm :wink:

MartijnBraam, 2022-03-14:
Exactly, Tow-Boot is the better solution for ppp booting and it already includes the JumpDrive functionality with the volume button when booting, making it a way neater solution in that case.

Danct12, 2022-03-14:
I also agree with Martijn Braam, it's the reason why Jumpdrive hasn't ported to the PinePhone Pro.

megous, 2022-08-14:
Well, now there's this: https://xff.cz/kernels/pinephone-pro-recovery/ which has a bit more potential than Tow-Boot for recovery over USB purposes and works in more situations.

MartijnBraam, 2022-08-15:
Can you please stop fragmenting the ecosystem for a bit

mburns9, 2022-08-22:
A dist-upgrade command in mobian last night and today, August 22, just removed the setup app and the pinephonepro-tweaks package in favor of the mobile-tweaks package. I need to restore the image saved from last afternoon.

PureTryOut, 2022-08-22:
I'm not sure what that has to do with Jumpdrive, please report it to the proper Mobian channels instead.