dreemurrs-embedded/Jumpdrive #79: Include luks support in jumpdrive

pravi, 2022-08-15:
Librem 5 support luks encrypted partitions but currently jumpdrive does not include support for mounting such partitions. Please include it. telneting to jumpdrive should be enough to use luks commands to mount the encrypted partitions.

Current work around is using the disk directly from host system but it needs qemu to mount the arm64 chroot.

MartijnBraam, 2022-08-15:
You don't need ARM64 to mount the luks partition on the host. If your host has the luks tools it should just work

pravi, 2022-08-15:
I can mount, but chroot needs qemu.

pravi, 2022-10-24:
I have collected my notes for using qemu to chroot into the librem 5 here https://people.debian.org/~praveen/librem5-rescue.txt

pravi, 2022-11-14:
Though using chroot via qemu is not sufficient to update the kernel on Librem 5 (this could be a bug in how crypttab is setup in pureos too)

```
Processing triggers for initramfs-tools (0.142) ...
update-initramfs: Generating /boot/initrd.img-6.0.0-1-librem5
W: No zstd in /usr/bin:/sbin:/bin, using gzip
cryptsetup: WARNING: target 'luks-802006bb-40ad-4973-bcbd-f6ac5a3cc093' not 
    found in /etc/crypttab
cryptsetup: ERROR: crypt_root: Source mismatch
device-mapper: table ioctl on crypt_root  failed: No such device or address
Command failed.
cryptsetup: WARNING: Couldn't determine cipher modules to load for crypt_root
Unsupported platform ''.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned erro

```

pravi, 2022-11-14:
After updating /etc/crypttab on the laptop, the error changes to

```
Processing triggers for initramfs-tools (0.142) ...
update-initramfs: Generating /boot/initrd.img-6.0.0-1-librem5
W: No zstd in /usr/bin:/sbin:/bin, using gzip
Unsupported platform ''.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit statu

```

This could probably be fixed in initramfs-tools or setting the value of platform where the tool is expecting.

pravi, 2022-11-14:
This may be related https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=951198

pravi, 2022-12-21:
On mobian, the kernel packages don't use flash kernel, so it can be updated using the chroot, though it is not yet relevant as mobian don't support luks yet. The steps for using qemu is documented at https://wiki.mobian.org/doku.php?id=tweaks#access-chroot-using-qemu-and-jumpdrive

pravi, 2022-12-22:
Also discussing the possibility of following mobian approach of kernel updates in pureos https://source.puri.sm/Librem5/OS-issues/-/issues/306